vovagorodok/ArduinoBleOTA #7: Incorrect firmware size

Gopijackson, 2023-04-04:
Upload to existing file esp32 completed successfully.
Added the led blinking program to the main file and then some errors in uploading the firmware.bin file to the esp32.
Incorrect firmware size error.
How to resolve this error

vovagorodok, 2023-04-04:
Are you using latest ArduinoBleOTA version?
What size of bin file is?
Upload by python script or mobile application?
if python script, can you show me console logs?

Gopijackson, 2023-04-05:
*ArduinoBleOTA version - v1
*bin file size  - 487.7 kB
* python script

*OTA Device Found!
Connecting to ArduinoBleOTA
HW: Example ESP32, VER: [1, 0, 0], SW: Example SW, VER: [1, 0, 0]
Begin upload: attr size: 128, buffer size: 5080
Uploaded: 127/487696
                 :
Uploaded: 487696/487696
Incorrect firmware size

vovagorodok, 2023-04-05:
Thanks
From logs I see that  you have outdated version. Please type git pull, will be easier work in same code base.
From bin size I see that default `ArduinoBLE` library is used.
Issue happens because of internal buffer to big and ESP32 wasn't able to handle fast incoming packages:
https://github.com/vovagorodok/ArduinoBleOTA/blob/main/doc/ADVANCED.md
I have tested mostly on `NimBLE-Arduino` library. Perhaps for `ArduinoBLE` library buffer size should be tuned better here:
https://github.com/vovagorodok/ArduinoBleOTA/blob/main/src/BleOtaSizes.h
I have decreased buffer for `ArduinoBLE`: https://github.com/vovagorodok/ArduinoBleOTA/commit/a808aead33b4b0a080da2bb6d1e24dbca68a49f6

You can run uploader.py second time. `ArduinoBleOTA` disables buffer when first upload fails. It should upload successfully but slowly.

You can disable buffer at all by:
```
build_flags =
	-D BLE_OTA_NO_BUFFER
``` 
But for `NimBLE-Arduino` speed decreases from 12kB/s to 3kB/s.
For `ArduinoBLE`  from 20kB/s to 3kB/s.

Gopijackson, 2023-04-06:
I am trying but can't solve this problem

 incorrect firmware size

Connecting to ArduinoBleOTA
HW: Example ESP32, VER:[1, 0, 0], SW: Example SW, VER:[1, 0, 0]
Begin upload: attr size: 256, buffer size:1020
org.bluez.Error.Failed: Failed to initiate write

vovagorodok, 2023-04-06:
attr size: 256 for `ArduinoBLE` is to big.That why you can't write to this characteristic.
Fix is on never version of ArduinoBleOTA, please:
```
git reset --hard
git pull
```
Add your blinking example main.ino and check if it works

Gopijackson, 2023-04-07:
espdev i tried this
attr size     : 16
buffer size : 512        perfectly flashed.But, very slow.
Then i tried this esp32dev_nimble, its perfectly flashed and time also ok.

Thank you for replying.

closed 2023-04-07 by Gopijackson