vovagorodok/ArduinoBleOTA #8: second service characteristics

Pikokosan, 2023-04-20:
How do you added characteristics to the secondary service?

I have a project i am working on and want to add the ble ota method you have built but i dont see how to add the characteristics to the second service. i assume you can use the standard nimble code but would just like to make sure.

arduino example would be great.

Pikokosan, 2023-04-20:
The existing example doesn't show how to reference the server.

Nimble example show
NimBLEService* pDeadService = pServer->createService("DEAD");

I know it's not pServer in your library but I'm not seeing how to point to the libraries server

vovagorodok, 2023-04-20:
You can call createServer a lot of times, because:
```
/* STATIC */ NimBLEServer* NimBLEDevice::createServer() {
    if(NimBLEDevice::m_pServer == nullptr) {
        NimBLEDevice::m_pServer = new NimBLEServer();
        ble_gatts_reset();
        ble_svc_gap_init();
        ble_svc_gatt_init();
    }

    return m_pServer;
} // createServer
```
Multiservice example updated. Please check and let me know i something should be added

Pikokosan, 2023-04-20:
Perfect. I'll give it a try when I get back to my computer. Thanks for the quick response

Pikokosan, 2023-04-21:
No go sir. kernel panic on boot. 

debug log looks to be not starting the second service.( 00c1acd4-f35b-4b5f-868d-36e5668d0929 for testing)
`entry 0x400805f0
[    19][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
D NimBLEServer: >> createService - 00c1acd4-f35b-4b5f-868d-36e5668d0929
D NimBLEServer: << createService
D NimBLEDevice: >> setLocalMTU: 527
D NimBLEDevice: << setLocalMTU
D NimBLEServer: >> createService - 15c155ca-36c5-11ed-adc0-9741d6a72f04
D NimBLEServer: << createService
D NimBLECharacteristic: >> setValue: length=3, data=000000, characteristic UUID=15c157fa-36c5-11ed-adc4-579c60267b47
D NimBLECharacteristic: << setValue
D NimBLECharacteristic: >> setValue: length=3, data=000000, characteristic UUID=15c1591c-36c5-11ed-adc6-dbe9603dbf19
D NimBLECharacteristic: << setValue
D NimBLEService: >> start(): Starting service: UUID: 15c155ca-36c5-11ed-adc0-9741d6a72f04, handle: 0xffff
D NimBLEService: Adding 6 characteristics for service UUID: 15c155ca-36c5-11ed-adc0-9741d6a72f04, handle: 0xffff
D NimBLEService: << start()
D NimBLEAdvertising: >> setAdvertisementData
D NimBLEAdvertising: << setAdvertisementData
D NimBLEAdvertising: >> setScanResponseData
D NimBLEAdvertising: << setScanResponseData
D NimBLEAdvertising: >> Advertising start: customAdvData: 1, customScanResponseData: 1
primary service
           uuid 0x1800
         handle 1
     end_handle 5
characteristic
           uuid 0x2a00
     def_handle 2
     val_handle 3
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2a01
     def_handle 4
     val_handle 5
   min_key_size 0
          flags [READ]
primary service
           uuid 0x1801
         handle 6
     end_handle 9
characteristic
           uuid 0x2a05
     def_handle 7
     val_handle 8
   min_key_size 0
          flags [INDICATE]
ccc descriptor
           uuid 0x2902
         handle 9
   min_key_size 0
          flags [READ|WRITE]
primary service
           uuid 15c155ca-36c5-11ed-adc0-9741d6a72f04
         handle 10
     end_handle 23
characteristic
           uuid 15c1564c-36c5-11ed-adc1-a3d6cf5cc2a4
     def_handle 11
     val_handle 12
   min_key_size 0
          flags [WRITE_NO_RSP]
characteristic
           uuid 15c156e2-36c5-11ed-adc2-7396d4fd413a
     def_handle 13
     val_handle 14
   min_key_size 0
          flags [READ|NOTIFY]
ccc descriptor
           uuid 0x2902
         handle 15
   min_key_size 0
          flags [READ|WRITE]
characteristic
           uuid 15c1576e-36c5-11ed-adc3-8799895de51e
     def_handle 16
     val_handle 17
   min_key_size 0
          flags [READ]
characteristic
           uuid 15c15886-36c5-11ed-adc5-1bc0d0a6069d
     def_handle 18
     val_handle 19
   min_key_size 0
          flags [READ]
characteristic
           uuid 15c157fa-36c5-11ed-adc4-579c60267b47
     def_handle 20
     val_handle 21
   min_key_size 0
          flags [READ]
characteristic
           uuid 15c1591c-36c5-11ed-adc6-dbe9603dbf19
     def_handle 22
     val_handle 23
   min_key_size 0
          flags [READ]

abort() was called at PC 0x400d4a8c on core 1

Backtrace: 0x40083791:0x3ffc9b70 0x40091d79:0x3ffc9b90 0x40097709:0x3ffc9bb0 0x400d4a8c:0x3ffc9c30 0x400d2861:0x3ffc9c70 0x400d1ace:0x3ffc9ca0 0x400d1bdd:0x3ffc9d10 0x400e52ae:0x3ffc9d90
`

vovagorodok, 2023-04-21:
NimBLE-Arduino requires `service->start();` for each service. Example updated

Pikokosan, 2023-04-25:
This worked and it is now advertising the second service correctly.

vovagorodok, 2023-04-25:
Should we close issue or something should be added/changed in examples?

Pikokosan, 2023-04-25:
in the bleotamultiservice.h change
`inline bool advertiseBle(const std::string& deviceName,
                         const std::string& primaryUUID,
                         const std::string& secondaryUUID)
{
    auto* server = BLEDevice::createServer();
    auto* advertising = server->getAdvertising();
}
`
to
`inline bool advertiseBle(const std::string& deviceName,
                         const std::string& primaryUUID,
                         const std::string& secondaryUUID)
{
    auto* server = NimBLEDevice::getServer();
    auto* advertising = server->getAdvertising();

}
`

in example you may add
`
#define CHARACTERISTIC_UUID_SECOND
`
`
#ifndef USE_NIM_BLE_ARDUINO_LIB
  static BLEService service(MY_SECOND_SERVICE_UUID);
  BLE.addService(service);
#else
  auto* server = BLEDevice::createServer();
  auto* service = server->createService(MY_SECOND_SERVICE_UUID);

    auto* pSecondCharacteristic = pService->createCharacteristic(
    CHARACTERISTIC_UUID_SECOND,
    NIMBLE_PROPERTY::WRITE | NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY);
  service->start();
#endif`

vovagorodok, 2023-04-26:
I think createServer is more safety.  According to characteristics in example, user of BLE library know how to do it

closed 2023-04-28 by vovagorodok