vovagorodok/ArduinoBleOTA #13: where do all the includes in BleOtaStorage.h come from ?

sdetweil, 2023-05-07:
I don't see an InternalStorageESP.h, just trying to build using examples/basic/main.ino

```
In file included from /Users/sam/Documents/Arduino/libraries/ArduinoBleOTA-main/src/ArduinoBleOtaClass.h:2,
                 from /Users/sam/Documents/Arduino/libraries/ArduinoBleOTA-main/src/ArduinoBleOTA.h:6,
                 from /Users/sam/Documents/Arduino/test_ota/test_ota.ino:1:
/Users/sam/Documents/Arduino/libraries/ArduinoBleOTA-main/src/BleOtaStorage.h:13:10: fatal error: InternalStorageESP.h: No such file or directory
 #include "InternalStorageESP.h"
          ^~~~~~~~~~~~~~~~~~~~~~
```

vovagorodok, 2023-05-07:
Its from `jandrassy/ArduinoOTA@^1.0.9`, that ArduinoBleOTA depend in

vovagorodok, 2023-05-07:
What are you using? And what I should add in library in order to see dependencies?

sdetweil, 2023-05-07:
I am using Arduino IDE

so I had to download each lib as a zip and add as library..

I am trying to add ESP32 non nimble to your lib, as I can't get Nimble to work

sdetweil, 2023-05-07:
so a comment in the doc with the github links to  dependencies if not using platformio

sdetweil, 2023-05-08:
using platfrormio now

closed 2023-05-08 by sdetweil