vovagorodok/ArduinoBleOTA #14: building w platformio on mac get this error

sdetweil, 2023-05-07:
Error: Could not find the package with 'vovagorodok/ArduinoBleOTA' requirements for your system 'darwin_arm64'

vovagorodok, 2023-05-07:
It still under development and ArduinoBleOta is not registered in platformio yet.
I'll register it after all tests.
Currently you can add full path:
```
https://github.com/vovagorodok/ArduinoBleOta
```
instead of:
```
vovagorodok/ArduinoBleOTA
```
And remove full path when ArduinoBleOTA I register it in platformio

vovagorodok, 2023-05-07:
Remember to add dependencies:
```
jandrassy/ArduinoOTA@^1.0.9
bakercp/CRC32@^2.0.0
h2zero/NimBLE-Arduino@^1.4.1
rlogiacco/CircularBuffer@^1.3.3

```

sdetweil, 2023-05-07:
yes, added, shame platformio doesn't do recursive deps.. anyhow.. now have a build failure in the ArduinoOTA package..

vovagorodok, 2023-05-07:
> platformio doesn't do recursive deps..

should have after adding [library.json](https://github.com/vovagorodok/ArduinoBleOTA/pull/11/files#diff-64cc7cdde6789dc1c1c4a6f406cd1ff6a6a00027097788e2d52fc27b0d75502e)
Will be added soon in https://github.com/vovagorodok/ArduinoBleOTA/pull/11

> build failure in the ArduinoOTA package..

Please show wrror

sdetweil, 2023-05-08:
thx, fixed that

closed 2023-05-08 by sdetweil