vovagorodok/ArduinoBleOTA #15: uploader.py/sh info

sdetweil, 2023-05-08:
can u provide any text of what is being accomplished?

I have the service swapped in now and can see its characteristics...

the bash script is looking two dirs up from tools folder, but on my mac under platformio its 4 levels up
and its looking at some json objects.. 

how is the hw/sw info used in the ESP on begin() compared to what on the sender app side? 
what is the HW 'name' and value supposed to represent? 

when platformio uploads the new code it reports
 ESP32-D0WD-V3 (revision v3.0)

I have to port all the uploader to my IOS config app. 
get info on images from some server api. then get the selected one (how tbd)  and then send to the connected device

sdetweil, 2023-05-09:
i have been unable to get python dbus, used by bluezero... on mac or ubuntu

sdetweil, 2023-05-09:
this made me update my 'app version's characteristic to retrieve the hw type. for image selection ...thx...

vovagorodok, 2023-05-09:
> the bash script is looking two dirs up from tools folder, but on my mac under platformio its 4 levels up
and its looking at some json objects..

Added more info in tools folder: https://github.com/vovagorodok/ArduinoBleOTA/blob/main/tools/README.md

> how is the hw/sw info used in the ESP on begin() compared to what on the sender app side?
what is the HW 'name' and value supposed to represent?

Added more information about it: https://github.com/vovagorodok/ArduinoBleOTA/blob/main/doc/VERSIONING.md

> when platformio uploads the new code it reports
ESP32-D0WD-V3 (revision v3.0)

It's not related to specific devkit hardware. HW name characteristic is used only for update possibilities. See `VERSIONING.md`

> I have to port all the uploader to my IOS config app.
get info on images from some server api. then get the selected one (how tbd) and then send to the connected device

You can check how it implemented in https://github.com/vovagorodok/ble_ota_app. It helps to small developers attach their devices to update system just by adding new line to this app: https://github.com/vovagorodok/ble_ota_app/blob/main/doc/ADD_NEW_HARDWARE.md and not create their new app for updating (if they want of course). Application is ready and will be released soon in Android and iOS stores.

sdetweil, 2023-05-09:
says dbus not found...

not a python guy. all the instructions for solving fail to build some 'wheel'?

sdetweil, 2023-05-09:
on the added Doc. got it..  I think I was expecting more checking against the info needed on begin(). 

thanks for app info.. not sure I can go that way was your service in not advertising unless my app  sends a trigger thru my service's characteristics. (ie, your service is ONLY available JUST for a the actual upload) 

app.. my app is written in Ionic, so there is a layer between my JS code and hardware access..  will see..

vovagorodok, 2023-05-09:
> says dbus not found...
> not a python guy. all the instructions for solving fail to build some 'wheel'?

In Ubuntu I just install bluezero:
```
pip3 install bluezero
```
Please show me error log

sdetweil, 2023-05-09:
ok dbus fixed... next 

```
(base) sam@sams:~/ArduinoBleOTA/tools$ pip3 install bluezero
Collecting bluezero
  Using cached bluezero-0.7.1-py2.py3-none-any.whl (41 kB)
Installing collected packages: bluezero
Successfully installed bluezero-0.7.1
(base) sam@sams:~/ArduinoBleOTA/tools$ ./uploader.py
Traceback (most recent call last):
  File "./uploader.py", line 2, in <module>
    from bluezero import adapter
  File "/home/sam/anaconda3/lib/python3.7/site-packages/bluezero/adapter.py", line 10, in <module>
    from bluezero import async_tools
  File "/home/sam/anaconda3/lib/python3.7/site-packages/bluezero/async_tools.py", line 7, in <module>
    from gi.repository import GLib
ModuleNotFoundError: No module named 'gi'
```
uname -a
Linux sams 5.4.0-122-generic #138-Ubuntu SMP Wed Jun 22 15:00:31 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal

vovagorodok, 2023-05-09:
https://stackoverflow.com/questions/21530577/fatal-error-python-h-no-such-file-or-directory

Try:
```
sudo apt install python3-dev
```

sdetweil, 2023-05-09:
thx, that worked, but no change really
```
(base) sam@sams:~/ArduinoBleOTA/tools$ pip3 install PyGObject  
Requirement already satisfied: PyGObject in /usr/lib/python3/dist-packages (3.36.0)
Collecting pycairo>=1.11.1
  Using cached pycairo-1.23.0.tar.gz (344 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: pycairo
  Building wheel for pycairo (PEP 517) ... done
  Created wheel for pycairo: filename=pycairo-1.23.0-cp38-cp38-linux_x86_64.whl size=325280 sha256=ca20052e697bf5f14f4e8f5d6cfe9bad8700872eed7a71a9d088648c31980aaf
  Stored in directory: /home/sam/.cache/pip/wheels/d9/25/86/ba7950c33840b1fda7ab932c7c7172d77b994807c5fd35c3ea
Successfully built pycairo
Installing collected packages: pycairo
Successfully installed pycairo-1.23.0
(base) sam@sams:~/ArduinoBleOTA/tools$ ./uploader.py
Traceback (most recent call last):
  File "./uploader.py", line 2, in <module>
    from bluezero import adapter
  File "/home/sam/anaconda3/lib/python3.7/site-packages/bluezero/adapter.py", line 10, in <module>
    from bluezero import async_tools
  File "/home/sam/anaconda3/lib/python3.7/site-packages/bluezero/async_tools.py", line 7, in <module>
    from gi.repository import GLib
ModuleNotFoundError: No module named 'gi'
```

vovagorodok, 2023-05-09:
https://stackoverflow.com/questions/71369726/no-module-named-gi

uploader.py i started from
```
#!/usr/bin/env python3
```

Maybe no the same instances.
Try `python3 uploader.py` instead `./uploader.py`

sdetweil, 2023-05-09:
same results

sdetweil, 2023-05-09:
closing now..

closed 2023-05-09 by sdetweil