vovagorodok/ArduinoBleOTA #17: uploader doesn't see service

sdetweil, 2023-05-09:
so I have swapped in your service,  and I can scan and find it, and connect and I can use the hw/sw chars to read from the service
so it is active 

disconnect and I still see it in multiple scanners with your service id 

but the uploader cannot see it 
I added a print after the scan 
```
dbus.Array([dbus.String('21010001-27b9-42f0-82aa-2e951747bbf9')], signature=dbus.Signature('s'), variant_level=1) -- second device
dbus.Array([dbus.String('11050001-27b9-42f0-82aa-2e951747bbf9')], signature=dbus.Signature('s'), variant_level=1)  -- original uuid
dbus.Array([dbus.String('0000fe50-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1)
dbus.Array([], signature=dbus.Signature('s'), variant_level=1)
dbus.Array([dbus.String('0000fe9a-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1)

dbus.Array([dbus.String('00000000-27b9-42f0-82aa-2e951747bbf9')], signature=dbus.Signature('s'), variant_level=1)--- this is the app

dbus.Array([], signature=dbus.Signature('s'), variant_level=1)
dbus.Array([], signature=dbus.Signature('s'), variant_level=1)
dbus.Array([dbus.String('0000fe50-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1)
dbus.Array([dbus.String('2e938fd0-6a61-11ed-a1eb-0242ac120002')], signature=dbus.Signature('s'), variant_level=1)
dbus.Array([], signature=dbus.Signature('s'), variant_level=1)
dbus.Array([dbus.String('0000110a-0000-1000-8000-00805f9b34fb'), dbus.String('0000110c-0000-1000-8000-00805f9b34fb'), dbus.String('0000110e-0000-1000-8000-00805f9b34fb'), dbus.String('00001200-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1)
dbus.Array([], signature=dbus.Signature('s'), variant_level=1)

```

sdetweil, 2023-05-09:
[image]

vovagorodok, 2023-05-09:
Not sure, but NimBle by default serve only one connected device. And if someone (some phone) is connected than no advertising at all. Try disconnect from all devices, maybe reboot device and try again. Remember that phone can try to reconnect automatically

sdetweil, 2023-05-10:
yes. once connected. cannot be seen by others.
I was just showing you the service description, which is good. and I can use the read only chars just fine.

so my server  is advertising and the service is good

so must be Linux BLE scanner...

vovagorodok, 2023-05-10:
Try to scan bluetooth devices from your OS directly instead uploader.py

sdetweil, 2023-05-10:
I used bleak to scan
and while the android scanner app sees your service and characteristics, the bleak scanner sees , the prior UUID service

```
c8:f0:9e:75:1a:26
C8:F0:9E:75:1A:26
{'path': '/org/bluez/hci0/dev_C8_F0_9E_75_1A_26', 'props': {'Address': 'C8:F0:9E:75:1A:26', 'AddressType': 'public', 'Name': 'c8:f0:9e:75:1a:26', 'Alias': 'c8:f0:9e:75:1a:26', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -47, 'Connected': False, 'UUIDs': ['11050001-27b9-42f0-82aa-2e951747bbf9'], 'Adapter': '/org/bluez/hci0', 'TxPower': 9, 'ServicesResolved': False}}
{'uuids': ['11050001-27b9-42f0-82aa-2e951747bbf9'], 'manufacturer_data': {}}
-46
ConfigApp
```
the app 
```python
import asyncio
from bleak import BleakScanner

async def main():
        devices = await BleakScanner.discover()
        for device in devices:
                print(device.name)
                print(device.address)
                print(device.details)
                print(device.metadata)
                print(device.rssi)

asyncio.run(main())
```

vovagorodok, 2023-05-10:
Can you show advertising data (not visible services after connect) in android nRF Connect application?

vovagorodok, 2023-05-10:
[image]

sdetweil, 2023-05-11:
[image]

sdetweil, 2023-05-11:
but connect and you see something different

[image]

vovagorodok, 2023-05-11:
In BLE is difference what services device have and what services device advertise.
Device can have unlimited number of services.
But advertising block is limited and you should chose what to advertise:
https://github.com/vovagorodok/ArduinoBleOTA/issues/10#issuecomment-1534745203
If device doesn't advertise ota service than script will not find him.

You can remove this filter if you want: https://github.com/vovagorodok/ArduinoBleOTA/blob/2e2a6857d72c1101b42b82a648cd13d142d74b26/tools/uploader.py#L76

sdetweil, 2023-05-11:
thx, i thought I handled that on the change. I don't see explicit remove (using BLE, see it in Nimble path)

am away til monday

sdetweil, 2023-05-11:
actually, that distinction might make this whole thing easier. setup the three services, and just decide which one to advertise.  my app discover handler just needs to know the found  advertised service's characteristics to use.(which it currently 'assumes').

vovagorodok, 2023-05-11:
But remember that it's not safety from security reason. Someone can connect and see that OTA services are available and flash other software.
You can additionally disable uploads and enable only when needed:
https://github.com/vovagorodok/ArduinoBleOTA/blob/main/doc/SECURITY.md#disable-uploads

sdetweil, 2023-05-11:
would only be available when my app says upload. not all the time.

sdetweil, 2023-05-12:
so I couldn't wait til Sunday and had some time. bought another esp32, and redid the code.

works as you said.  what's advertised vs services

[image]
[image]

sdetweil, 2023-05-12:
successful connect and upload!!!

non-nimble!

sdetweil, 2023-05-12:
only 764b/s, 24 min 17 sec for 1113712 bytes

doesn't reboot?
I have to do that?

sdetweil, 2023-05-12:
so I think I should close this as it was my fault, and fixed.  and move to another topic about the finish up.  apply/reboot...

vovagorodok, 2023-05-13:
> so I think I should close this as it was my fault, and fixed. and move to another topic about the finish up. apply/reboot...

Perhaps you forgot `ArduinoBleOTA.pull();` in `loop()`

sdetweil, 2023-05-14:
well, now my system has cached (across boots)  the device uuids and the upload service is not visible. 
I redid the startup to only have ONE service, and the scan data still has both.

sdetweil, 2023-05-16:
ok, figured out how to clear the cache, now get an upload failure  using NimBLE
```
connected  -- my app
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
on disconnect start advertising= 15c155ca-36c5-11ed-adc0-9741d6a72f04 -- my app
disconnected -- my app 
```

sdetweil, 2023-05-16:
using non-nimble ESP32 BLE.

Installing. Upload time: 0:05:07.499808
Success!
(base) sam@sams:~/ArduinoBleOTA/tools$ rexx rexxtry say "1138176/((5*60)+07)"
3707.41368

sdetweil, 2023-05-18:
to clear cache
stop bluetooth service
delete the cache files (on linux/var/lib/bluetooth)  under local device entry/folder)
restart bluetooth service
then can scan 
have to do this all the time. 

will provide the script (bash) code for this..  (need to test on mac)

sdetweil, 2023-05-24:
got nimble working, 
Uploaded: 1139200/1139200
Installing. Upload time: 0:01:23.745954

"(1139200/(1*60+23))"

13725/bytes sec

the Nimble author said not to server->addService()

sdetweil, 2023-06-21:
I think I can close this now

closed 2023-06-21 by sdetweil