vovagorodok/ArduinoBleOTA #20: remove advertising functionality from begin methods

sdetweil, 2023-05-18:
in my case I know when to advertise, ,  you should leave that up to the user.

this sort of goes with the MTU issue

vovagorodok, 2023-05-18:
> in my case I know when to advertise, , you should leave that up to the user.

That why we have two begin methods. First is with ble initialization and advertising for basic users. But second is without for advanced users and multiservice

> this sort of goes with the MTU issue, https://github.com/vovagorodok/ArduinoBleOTA/issues/19

What do you mean?

sdetweil, 2023-05-18:
but one inits the BLE system with a name.. no thanks.. and advertisises.. no thanks

```
    BLEDevice::init(deviceName);
    auto* server = BLEDevice::createServer();

    if(!begin(storage, hwName, hwVersion, swName, swVersion, enableUpload))
        return false;

    auto* advertising = server->getAdvertising();
    advertising->setScanResponse(true);
    advertising->setMinPreferred(0x06); // functions that help with iPhone connections issue
    advertising->setMaxPreferred(0x12);
    return advertising->start();
```

the other creates the server .. no thanks 
and adds the service to the system..  no thanks..
and adds it to the advertisable data
```
 auto* server = BLEDevice::createServer();
    BLEDevice::setMTU(BLE_OTA_MTU_SIZE);

    bleOtaUploader.begin(storage);
    bleOtaUploader.setEnabling(enableUpload);
    auto* service = server->createService(BLE_OTA_SERVICE_UUID);

    auto* rxCharacteristic = service->createCharacteristic(
        BLE_OTA_CHARACTERISTIC_UUID_RX,
        NIMBLE_PROPERTY::WRITE_NR
    );
    rxCharacteristic->setCallbacks(this);

    auto* txCharacteristic = service->createCharacteristic(
        BLE_OTA_CHARACTERISTIC_UUID_TX,
        NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::NOTIFY
    );
    this->txCharacteristic = txCharacteristic;

    begin(*service, hwName, hwVersion, swName, swVersion);

    auto* advertising = server->getAdvertising();
    advertising->addServiceUUID(BLE_OTA_SERVICE_UUID);
    return service->start();
```

i extended the second, pass in the server, get back the service

but I see that advertising side effect..  

I have a problem with nimble when calling begin during loop()..  i get an empty service back.. no characteristics..
I 'think' the storage used is not permanent...   works ok at setup() time..  but adds it to the services list and starts it.. 
i don't want to expose the OTA service until its time to DO OTA to a specific device..  and thats my app/user
 decision..

vovagorodok, 2023-06-17:
But Your case looks very specific. Most of users will prefer simple one/two line integration I guess.

sdetweil, 2023-06-17:
I understand. just trying to implement some reasonable security.  
and still be in control of what's happening.

closed 2023-11-15 by vovagorodok