vovagorodok/ArduinoBleOTA #25: enable/disable upload feature understanding

sdetweil, 2023-06-20:
so, I want to  disable upload until a later time. 

I call begin(.... enableUpload:false)

later I want to enable it

so I call
  ArduinoBleOTA.enableUpload();

but my BEGIN transmission get a status 0x06 back, 
#define UPLOAD_DISABLED 0x06

if I change the begin(.... enableUpload:true)
then my BEGIN transmission is accepted

am I using it the right way?

vovagorodok, 2023-06-22:
Hmm,
begin(.... enableUpload:false) and than ArduinoBleOTA.enableUpload(); should work. Ill check it

vovagorodok, 2023-08-07:
Checked. Works correctly begin(.... enableUpload:false) and than ArduinoBleOTA.enableUpload();. It only changes one boolean flag, can't understand where can be issue from Your side. Do you have latest ArduinoBleOTA version?

closed 2023-11-15 by vovagorodok