vovagorodok/ArduinoBleOTA #27: which crc lib?

sdetweil, 2023-06-24:
I am looking for a working crc for nodejs app.

the one I have produces different results with running crc like python code and one time like flutter app.
and different than bleota code version

vovagorodok, 2023-06-24:
BleOta uses default CRC32 calculation. Find nodejs library with that

sdetweil, 2023-06-24:
thanks. first two produce different results...

vovagorodok, 2023-06-24:
Hmm, am I understand correctly uploader.sh and flutter app has different result for same binary?

sdetweil, 2023-06-24:
>correctly uploader.sh and flutter app has different crc result for same binary
no, you just use two different approaches,   flutter is from one buffer, uploader is incremental 

man this is harder than it should be

download the same file from the same server, wrote a node test app 

same crc as your lib
your lib gets the data from my app (so that is good) 
my app calculates from full data (like you flutter app)  like the node test app 

node test app 226795867 
your lib 226795867 

my app BEFORE sending the data 3074055154, technically the crc lib returns signed number -1220912142 

finally had to embed in source 
https://stackoverflow.com/questions/18638900/javascript-crc32

sdetweil, 2023-06-25:
working

closed 2023-06-25 by sdetweil