vovagorodok/ArduinoBleOTA #36: Security pin

faizannazir, 2024-03-06:
what is security pin and how can we set our own i tried  ota_pin as pin 
 and tried setpin function to set both didnot worked  for me

vovagorodok, 2024-03-06:
Here is small doc about security: https://github.com/vovagorodok/ArduinoBleOTA/blob/main/doc/SECURITY.md

From doc You can see that there are two options to provide a security by disabling/enabling uploads (using button/switch for example) or set a pin code. See:
[image]

vovagorodok, 2024-03-06:
because You set pin code before

faizannazir, 2024-03-06:
> because You set pin code before

by security.setpin(12345) ???

without this code also it ask for pin  and  pin 12345 also not work

vovagorodok, 2024-03-06:
As I remember Nim-Ble stack requires exact 6 digits pin. Try 123450 or 012345

Proposed to use ble_ota_app to change pin

vovagorodok, 2024-03-06:
Renamed a bit from `pin` to `pin code`

faizannazir, 2024-03-07:
how to run these examples on arduino ide is there any possibility to make them compatible with arduino ide

vovagorodok, 2024-03-07:
I'm PlatformIO user, for me easier to just add to my project:
```
build_flags =
	-D USE_NIM_BLE_ARDUINO_LIB
lib_deps =
	https://github.com/vovagorodok/ArduinoBleOTA
```
I have plans to add this library to PlatformIO Hub and prepare it to work in Arduino IDE.
But for this moment there is one small issue that should be solved with non NimBle part of this library.
I hope this month I'll done all of that

vovagorodok, 2024-03-07:
Created issue in order to remember about it: https://github.com/vovagorodok/ArduinoBleOTA/issues/37

vovagorodok, 2024-03-19:
Library available at ArduinoIDE as vovagorodok_ArduinoBleOTA.\
And at PlatformIO as:
```
build_flags =
	-D USE_NIM_BLE_ARDUINO_LIB
lib_deps =
	vovagorodok/ArduinoBleOTA@^1.0.0
```

closed 2024-03-25 by faizannazir

faizannazir, 2024-05-30:
> Library available at ArduinoIDE as vovagorodok_ArduinoBleOTA. And at PlatformIO as:
> 
> ```
> build_flags =
> 	-D USE_NIM_BLE_ARDUINO_LIB
> lib_deps =
> 	vovagorodok/ArduinoBleOTA@^1.0.0
> ```

Arduino ide generated bin file new version not updated using ble app it show 100 % but software not updated 

flashed program 
```
`#include <ArduinoBleOTA.h>

#define DEVICE_NAME "Esp32 Ble"
#define HW_NAME "Esp32 HW"
#define HW_VER {1, 0, 0}
#define SW_NAME "Esp32 SW"
#define SW_VER {1, 0, 0}

#define BUZZER_PIN 15  // Buzzer Pin

unsigned long previousMillis = 0;
const long interval = 1000; // Interval for buzzering
bool isBuzzing = false;

void setup() {
  Serial.begin(9600);
  ArduinoBleOTA.begin(DEVICE_NAME, InternalStorage, HW_NAME, HW_VER, SW_NAME, SW_VER);
  Serial.println("Buzzer test");
  pinMode(BUZZER_PIN, OUTPUT);
}

void loop() {
  BLE.poll();

  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;

    if (isBuzzing) {
      noTone(BUZZER_PIN);
      Serial.println("Buzzering stop");
    } else {
      tone(BUZZER_PIN, 1000);
      Serial.println("Buzzering");
    }

    isBuzzing = !isBuzzing; // Toggle the buzzing state
  }
}
`
```

updating over ble new program 

```

#include <ArduinoBleOTA.h>
#define DEVICE_NAME "Esp32 Ble"
#define HW_NAME "Esp32 HW"
#define HW_VER {1,0, 0}
#define SW_NAME "Esp32 SW"
#define SW_VER {1, 1, 0}

#define RELAYPIN 4

unsigned long previousMillis = 0;
const long interval = 1000; // Interval for Relay
bool isRelayON = false;

void setup() {
  Serial.begin(9600);
  ArduinoBleOTA.begin(DEVICE_NAME, InternalStorage, HW_NAME, HW_VER, SW_NAME, SW_VER);
  Serial.println("Relay test");
  pinMode(RELAYPIN, OUTPUT);
}

void loop() {
  BLE.poll();
    unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;

    if (isRelayON) {
      digitalWrite(RELAYPIN,LOW);
      Serial.println("Relay OFF");
    } else {
      digitalWrite(RELAYPIN,HIGH);
      Serial.println("Relay ON");
    }

    isRelayON = !isRelayON; // Toggle the Relay state
  }
}

```

the bin file i used is programname.ino.bin