vovagorodok/ArduinoBleOTA #40: Example of combining this with lets say a BLE Serial application

dBitech, 2024-06-08:
Would be nice to see some expanded examples in the "multi service" area.

vovagorodok, 2024-06-08:
Hmm, I'll think about this,
For this moment please take a look on draft blink example based on: https://github.com/arduino-libraries/ArduinoBLE/blob/master/examples/Peripheral/LED/LED.ino

```
#include <ArduinoBLE.h>
#include <ArduinoBleOTA.h>
#include <BleOtaMultiservice.h>

#define DEVICE_NAME "BLE LED OTA"
#define LED_SERVICE_UUID "19B10000-E8F2-537E-4F6C-D104768A1214"
#define LED_CHARACTERISTIC_UUID "19B10001-E8F2-537E-4F6C-D104768A1214"

#define LED_BUILTIN 2

BLEService ledService(LED_SERVICE_UUID); // Bluetooth Low Energy LED Service
BLEByteCharacteristic switchCharacteristic(LED_CHARACTERISTIC_UUID, BLERead | BLEWrite);

const int ledPin = LED_BUILTIN;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  pinMode(ledPin, OUTPUT);

  if (!initBle(DEVICE_NAME)) {
    Serial.println("starting Bluetooth Low Energy module failed!");
    while (1);
  }

  ledService.addCharacteristic(switchCharacteristic);
  BLE.addService(ledService);

  // set the initial value for the characeristic:
  switchCharacteristic.writeValue(0);

  ArduinoBleOTA.begin(InternalStorage);
  advertiseBle(DEVICE_NAME, LED_SERVICE_UUID);

  Serial.println("BLE LED Peripheral");
}

void loop() {
  BLEDevice central = BLE.central();

  if (central) {
    Serial.print("Connected to central: ");
    Serial.println(central.address());

    while (central.connected()) {
      if (switchCharacteristic.written()) {
        if (switchCharacteristic.value()) {
          Serial.println("LED on");
          digitalWrite(ledPin, HIGH);
        } else {
          Serial.println(F("LED off"));
          digitalWrite(ledPin, LOW);
        }
      }
      ArduinoBleOTA.pull();
    }

    Serial.print(F("Disconnected from central: "));
    Serial.println(central.address());
  }
}
```

You can use `nRF Connect` android application. And than:
1. Find and connect to device via `nRF Connect`
2. Click to LED service `19B10000-E8F2-537E-4F6C-D104768A1214`
3. Write `0x00` and `0x01` to the LED characteristic `19B10001-E8F2-537E-4F6C-D104768A1214`

You sholud see ESP32 blinking LED

vovagorodok, 2024-06-08:
Here is another example: https://github.com/TimoKropp/OPENCHESSBOARD_BLE/blob/main/src/main.ino