vovagorodok/ArduinoBleOTA #43: Immediately disconnected

phhoef, 2025-02-05:
I am trying to implement over the air update via BT to my ESP32 (esp-s3-devkitc-1).
I see the published service, when trying to connect it takes several seconds and the disconnects.
I am using platform.io with Arduino framework.

```
python3 uploader.py ./.pio/build/ota/firmware.bin
Devices:
0. [CDF385A9-A027-1041-DC54-D21216A2885D2] ArduinoBleOTA
Chose device [0]: 0
Connecting to ArduinoBleOTA
disconnected
```

In my setup method I call a task for ota
```c++
xTaskCreate(ota, "OTA", 6000, NULL, 4, NULL);
```

```c++

#include "freertos/task.h"
#include "ArduinoBleOTA.h"

#define DEVICE_NAME "ArduinoBleOTA"
#define MF_NAME "Example MF"
#define HW_NAME "Example ESP32"

#define HW_VER {1, 0, 0}
#define SW_NAME "Example SW"
#define SW_VER {1, 0, 0}

void ota(void *pvParameter)
{
    Serial.println("Starting BLE OTA Service...");

    if (!ArduinoBleOTA.begin(DEVICE_NAME,
                             InternalStorage,
                             MF_NAME,
                             HW_NAME,
                             HW_VER,
                             SW_NAME,
                             SW_VER))
    {
        Serial.println("Failed to init BLE OTA!");
        return;
    }

    Serial.println("BLE OTA Service started");

    while (1)
    {
        ArduinoBleOTA.pull();
        vTaskDelay(pdMS_TO_TICKS(10));
    }
}
```
Once the connection failed, the advertising disappeared.

I tried to connect directly with `Bleak`, but I do have the same behaviour.
That's why I started thinking, that it might be related to the library itself.

Does anybody have an idea what I am doing wrong?

vovagorodok, 2025-02-05:
Are You using `USE_NIM_BLE_ARDUINO_LIB` in platformio.ini?

phhoef, 2025-02-05:
Thanks for your fast reply and your help.
I also tried moving the stuff out of the task to the mainloop, but with the same result.

Anyhow, I am now trying to use NimBLE, but I have problems as `BLE` is not found.
```
src/tasks/ota.h:36:9: error: 'BLE' was not declared in this scope
         BLE.poll();
```
I tried including `ArduinoBLE.h` but then I get a lot of errors.
```
In file included from .pio/libdeps/esp32dev/ArduinoBleOTA/src/ArduinoBleOtaClassNimBle.h:6,
                 from .pio/libdeps/esp32dev/ArduinoBleOTA/src/ArduinoBleOTA.h:5,
                 from src/tasks/ota.h:5,
                 from src/main.h:12,
                 from src/main.cpp:1:
.pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEDevice.h:75:41: error: redefinition of 'class NimBLEDescriptor'
```

It feels like I am doing something fundamental wrong.

vovagorodok, 2025-02-05:
`BLE.poll()` is only for ArduinoBLE. NimBLE-Arduino doesn't require `BLE.poll()`

phhoef, 2025-02-05:
Ah sorry, that might be the original problem. Initially, I thought that BLE.poll() was only necessary with NimBLE.
I can't test it until tomorrow.

phhoef, 2025-02-06:
Thanks for your help.
In fact, `BLE.poll()` was missing for the ArduinoBLE lib. As I said, I was convinced that this is only necessary for NimBLE lib 🙈

It's working like a charm now 😄 Sorry for trouble ...

closed 2025-02-06 by phhoef