vovagorodok/ArduinoBleOTA #44: Firmware Upload broken w/ esp32dev_native build

lincomatic, 2025-03-20:
Using the bundled platformio.ini, I built full example by selecting env:esp32dev_nimble, and
I can upload firmware successfully via uploader.py.

But with native BLE build using env:esp32dev_native, uploader.py gives the following error:

```
Chose device [0]:
Connecting to ArduinoBleOTA
MF: Example MF, HW: Example ESP32, VER: [1, 0, 0], SW: Example SW, VER: [1, 0, 0]
[WinError -2140864509] The attribute cannot be written
```

vovagorodok, 2025-03-20:
Strange, can't reproduce env:esp32dev_native with Linux from my side. Perhaps it can be Windows issue. What OS do You have?

lincomatic, 2025-03-21:
> Strange, can't reproduce env:esp32dev_native with Linux from my side. Perhaps it can be Windows issue. What OS do You have?
So upload.py works for you with esp32dev_native?

I tested in Windows. OK, I will try again with Linux and see if I get a different result

lincomatic, 2025-03-21:
I tried on 4 different linux machines, but I can't find one that has BLE support, so I can't test on Linux :(
the esp32dev and esp32dev_nimble both work fine on my Windows machine. Only esp32dev_native fails

lincomatic, 2025-03-22:
OK,I tested with your Android app, and it works wit native build. So there's some sort of incompatibility with Windows 10/11. It's strange how ArduinoBle and nimble builds work OK w/ Windows.

vovagorodok, 2025-03-24:
Currently I haven't Windows machine to check it. I'll try to find and check